chipsalliance/fpga-interchange-schema #12: Property type considerations

gatecat, 2021-02-24:
At the moment, it appears like there are three allowed kinds of properties:
 - strings
 - 32-bit integers
 - booleans

The second two of these look reasonable. However, I think it would be prudent to distinguish between a string that is actually a string value (something like https://github.com/YosysHQ/yosys/blob/master/techlibs/nexus/cells_sim.v#L95, Lattice use these a lot) and a string value that is actually containing a bitvector (e.g. encoded as a Verilog constant).

Without that difference being stored; I don't think it will be possible to reliably go back to a Verilog netlist from an interchange netlist - the original value could have either been the bitvector `16'h1234`; or actually a string `"16'h1234"` (equivalent to a much longer bitvector containing the ASCII values of those characters).

Yosys originally made a similar design mistake in its JSON format and now we have to live with a somewhat horrible escaping scheme involving postfixed spaces for disambiguation: https://github.com/YosysHQ/yosys/blob/master/backends/json/json.cc#L403-L405

litghost, 2021-02-24:
String properties **require** quotes if they are quoted strings.  My processing of Yosys JSON might be messing this up.  So for example:

```
EX #(.PROP1("TRUE"), .PROP2(16'h1234)) ex();
```

Should actually be encoded as:
```
{
  .key = "PROP1"
  .textValue = "\"TRUE\""
},
{
  .key = "PROP2"
  .textValue = "16'h1234"
}
```

That make sense?

gatecat, 2021-02-24:
Yes, that makes sense. I hadn't actually checked that the JSON frontend was doing.

I'm still not sure if encoding two fundamentally different types of property into a string is the best option with access to all the power of a proper schema; storing bit vector parameters as a vector of bytes/32-bit words/etc would seem easier to work with to me.

litghost, 2021-02-24:
> storing bit vector parameters as a vector of bytes/32-bit words/etc would seem easier to work with to me.

This immediately runs into problems with LUTs (64-bits) and BRAM init strings (256-bits).  As a future extension, we could add a varint property field, but I think operating on verilog encoded strings is probably the right short-term answer.

gatecat, 2021-02-24:
> This immediately runs into problems with LUTs (64-bits) and BRAM init strings (256-bits). 

Yes, that's why I suggested a vector of bytes/words

> I think operating on verilog encoded strings is probably the right short-term answer.

Don't have a massive problem with this provided that the semantics are well defined so we don't end up in a situation where different tools have slightly different ideas of what Verilog encoded string means (ime these kinds of differences tend to cause big problems with EDIF interoperability).

litghost, 2021-02-24:
> Don't have a massive problem with this provided that the semantics are well defined so we don't end up in a situation where different tools have slightly different ideas of what Verilog encoded [...]

So the initial solution here is to **require** that user supplied primitive cell parameters follow the default parameters of the library definition!  So if the device database primitive cell default is `64'b0`, then it is **required** that the parameter use verilog *binary* bit strings.  If the default is `256'h0`, then it is **required** that use verilog *hex* bit strings.  This is something that I've been planning on writing up in a `docs` folder, but wanted to get a basic non-timing driven P&R up and running before doing so.

litghost, 2021-03-19:
> Don't have a massive problem with this provided that the semantics are well defined so we don't end up in a situation where different tools have slightly different ideas of what Verilog encoded [...]

I've tried to address this in #19, take a look with that in mind.

PR #19 (2021-03-22): Add entries for default parameters, and how to present parameters as strings.
Fixes #12 (from a schema perspective).
closed 2021-03-22 by litghost