chipsalliance/fpga-interchange-schema #49: Capnp Schema File Structure?

reillymck, 2021-04-30:
Hey, so I think I am missing something here. So after running make, I copy schema/capnp/java.capnp to /capnp/java.capnp. However, I get the following error:
`capnp.lib.capnp.KjException: ~/RapidWright/interchange/fpga-interchange-schema/interchange/DeviceResources.capnp:15: failed: Import failed: /capnp/java.capnp`

I ran `sudo chmod +777 /capnp` and `sudo chmod +777 /capnp/java.capnp` and the import still fails. As a temporary solution, I have edited the schema files to read `import ../../schema/capnp/java.capnp` instead, and this works. However, I would like to know how everyone else is structuring their directory since I am obviously missing something here.
Thanks.

gatecat, 2021-05-01:
Have you tried a `sudo make install` of `capnproto-java`? That is what our CI is doing: https://github.com/YosysHQ/nextpnr/blob/master/.github/ci/build_interchange.sh#L16

reillymck, 2021-05-24:
Thanks! This worked for me.

closed 2021-05-24 by reillymck